umd-memsys/DRAMSim2 #58: Meaning of ini parameters

dzaragoza, 2015-05-08:
Hello

Reading the example ini files I don't understand what the following parameters mean:

- AL
- BL
- tRTRS
- tCMD 

Could you please explain what those parameters mean?

Kind regards
David

dzaragoza, 2015-06-19:
To answer part of my own question: AL is the Additive Latency used in posted CAS and BL is the Burst Length. I'm guessing tCMD is the delay between commands, but I still don't know what tRTRS is.

dramninjasUMD, 2015-06-19:
tRTRS is the rank-to-rank switching time. Basically this is a delay needed to add some quiet time on the shared bus when we change who is currently driving the bus. 

tCMD is just how long it takes to transmit a command.

dramninjasUMD, 2015-06-19:
Just one more note. For anyone looking to get a good overview of DRAM timing parameters but don't have access to the memory book, Dave Wang's thesis is pretty stellar: http://www.ece.umd.edu/~blj/papers/thesis-PhD-wang--DRAM.pdf

closed 2015-06-19 by dramninjasUMD