umd-memsys/DRAMSim2 #59: DRAMSim2-Gem5 Average Latency

afinci, 2016-08-17:
Hi,

I am using your default ini files in my simulations (./DRAMSim -t traces/k6_aoe_02_short.trc -s system.ini.example  -d ini/DDR3_micron_32M_8B_x8_sg15.ini  -c 1000000). Average latency values are 280-300 ns. However, people use 50-100 ns as memory latency values in Gem5. It seems like my latency values are much higher than usual. I was wondering what the reason is for this difference between two latency values? 

Thanks in advance,

Ahmet

shavvn, 2016-09-28:
Hi Ahmet (@afinci),

We've moved DRAMSim2 to a new repo in https://github.com/umd-memsys. We will be providing support from there. 

In answering your question, I think the reason is that k6_aoe_02_short.trc is only an example trace. The average latency depends heavily on your modeled system configurations and the work load (memory access patterns). In some system setup you could indeed get latency in the smaller range. 

Best,
Shawn

afinci, 2016-09-28:
Hi @shavvn ,

Thanks for your answer. Actually, I understood the problem and tried to solve it by writing a program called "remover.py". Basically, it randomly removes 'x' percentage of lines of the given memory trace to obtain a relatively less memory intensive trace. I posted it to the project's repo in case people may want to use it.

Thanks,

Ahmet