umd-memsys/DRAMSim2 #67: trace files fro running DRAMSim2

naveen168, 2017-03-19:
I'm trying to get trace files to run the DRAMSim2 simulator. If anyone can provide some tracefiles other than the ones provided in the ZIP file, that would be helpful.
Thanks in advance.

shavvn, 2017-03-20:
Hi @naveen168 ,

The trace files there are sufficient for testing the simulator itself. What do you need more trace files for?

konbick, 2017-03-21:
The .k6 trace file, however, was created with the pintool that you mentioned in another issue. Is that correct?

naveen168, 2017-03-21:
Hi @shavvn 

Thanks for your response. I'm trying to implement a restore after a read scheme for my project, in which i need to have more than six trace files to test  my output. So if you have a documentation to create my own traces or if you have already generate those files, Please share them. It would be a great help.

shavvn, 2017-03-21:
@naveen168 like konbick mentioned, you can use Intel's pin tool to create memory reference traces. The output format might be slightly different from what DRAMSim2 takes, but it should be easy to convert them.

naveen168, 2017-03-22:
@shavvn Thanks for your response. I went through the documentation of the pin tool to create memory reference traces, but cannot understand how to introduce the clock cycles in the trace file. And also in the documentation they have generated trace files on performing bin/ls operation. Can we use  same operation for collecting traces or tool should be performed on running any benchmarks.
I really appreciate if can help me through this. Thanks in advance.

shavvn, 2017-03-22:
I did not look into how to put the clock information in the trace file yet but I'm sure it shouldn't be very hard to do so.

Also in DRAMSim2 you can choose to ignore cycle information, i.e. running as fast as it could. But I don't know if it would be still legit for your purpose.

they use bin/ls as an example. In theory, you can use pin to run any executable as long as they're executable when you're not using pin.

closed 2018-08-24 by shavvn