umd-memsys/DRAMSim2 #68: Would you be willing to share your validation flow?

davidbiancolin, 2017-06-22:
Hey, 

In section III of the CAL paper, you guys describe your validation procedure. Would you be willing to share your flow? The ruby script does not seem to be provided, and from I can tell, you've made at least some modifications to the Micron provided testbench, to support things like multi-rank organizations. I'd also like to know what mode register settings you are using.

shavvn, 2017-06-26:
Hi @davidbiancolin ,

Unfortunately we did not inherit the ruby scripts from the original developers either... but I figured out the validation process and here are some hints of how to do it (I might release another script of doing this later):

If you have micron's verilog model, you basically need to generate your own version of the "subtest.vh", and also figure out the command line for the simulation software you're using (modelsim, ncverilog, etc).

As for mode registers, make sure the timing-related parameters are set correctly (e.g. CL, tWR) and matches your input ini file.

davidbiancolin, 2017-06-26:
Hey @shavvn, 

Thanks for your reply! That is more or less what i'm doing. What i'm interested in particularly, is how you guys modified the testbench to support multi-rank validation, and what sort of termination schemes you were modelling. For example, the validation output of DRAMsim2 prepends the rank to the commands in subtest.vh, but the functions in the micron-provided testbench don't have an argument for rank. 

In any case, I've made modifications of my own to do this, but it would have been nice to have a sanity check.

closed 2017-06-26 by davidbiancolin