umd-memsys/DRAMSim2 #73: Generating physical address trace of a program

sarabjeetsingh007, 2018-07-09:
Hi,
I see that DRAMSim2 provides an in-depth simulation of the main memory, but requires a physical address trace of the program as the input. How to generate these physical address traces of an application, as most tools provide ways to generate virtual address traces, not physical.

shavvn, 2018-08-24:
You can use a CPU simulator that simulates a TLB (the unit that does the VA->PA translation) to generate physical address. For example, Gem5.

closed 2018-08-24 by shavvn

liuhy397, 2019-07-27:
> You can use a CPU simulator that simulates a TLB (the unit that does the VA->PA translation) to generate physical address. For example, Gem5.

dear shavvn,do you know how to generate a physical trace file with gpgpu-sim,thank you very much